mikeakohn/naken_asm #139: Display error disassembling PIC32

russells, 2025-01-10:
Disassembling some PIC32 code, I noticed that an address for a BEQ instruction (and others) was not completely disassembled. If I put an Offset(18) of 0x12340, the opcode is generated correctly, but displayed truncated as 0x2340.

Session below shows this:

```
$ ll
total 12
drwxr-xr-x 2 russells russells 4096 Jan 10 15:58 ./
drwxr-xr-x 3 russells russells 4096 Jan 10 15:58 ../
-rw-r--r-- 1 russells russells   30 Jan 10 15:56 beq.asm
$ cat beq.asm 
.pic32
BEQ $t0,$zero,0x12340
$ naken_asm beq.asm 

naken_asm

Authors: Michael Kohn
         Joe Davisson
    Web: https://www.mikekohn.net/
  Email: mike@mikekohn.net
Version: October 19, 2024

 Input file: beq.asm
Output file: out.hex

Pass 1...
Pass 2...

Program Info:
Include Paths: .
               /usr/local/share/naken_asm/include
               include
 Instructions: 1
   Code Bytes: 4
   Data Bytes: 0
  Low Address: 0x0000 (0)
 High Address: 0x0003 (3)

$ cat out.hex
:04000000CF480011D4
:00000001FF
$ naken_util -pic32 out.hex 

naken_util - by Michael Kohn
                Joe Davisson
    Web: https://www.mikekohn.net/
  Email: mike@mikekohn.net

Version: October 19, 2024

Loaded out.hex of type hex / pic32 from 0x0000 to 0x0003
Type help for a list of commands.
stopped> disasm

Addr    Opcode Instruction                              Cycles
------- ------ ----------------------------------       ------
0x0000: 0x110048cf beq $t0, $0, 0x2340 (offset=9020)        1
stopped> quit
$ 
```

The opcode 0x110048cf is `000100 (opcode) 01000 (R8=$t0) 00000 (R0) 0100 1000 1100 1111 (Offset18>>2)`

0x48cf shifted right two bits is 0x1233C, plus four to make it relative to the instruction is 0x12340.

So I'm sure the binary is right, but the disassembly is incorrect.

russells, 2025-01-10:
The other instructions I mentioned included at least the BNE instruction.  I can test more if you need.

mikeakohn, 2025-01-10:
Hmmm.. I think this actually might be an assembler "issue". The branch instructions in MIPS are 16 bit signed and 0x12340 overflows that. It seems like the assembler should have produced an error warning the branch is too far?

mikeakohn, 2025-01-10:
Thanks for letting me know about this btw... it's helpful.

mikeakohn, 2025-01-12:
It was in the disassembler rather than the assembler. Should be fixed now. Thanks!

closed 2025-01-12 by mikeakohn

russells, 2025-01-16:
Thank you for looking at this.  Works now.