mikeakohn/naken_asm #140: Macro parameters - remove the parentheses?

russells, 2025-01-16:
This is more a feature request than an issue.

I'm writing PIC32 (MIPS RISC) code, and I have a request for how macros are called.

PIC32, like many RISC processors, has a delay slot after a branch. The delay slot is intended for use by compilers, but I want to ignore it for now since I'm writing the assembler code myself.  So I'm writing code like:
```
bne $zero, $t1, there
nop
```
I wrote a macro (actually one for each branch instruction) like this:
```
.macro _bne(r1,r2,x)
bne r1,r2,x
nop
.endm
```
and it's called like this:
```
_bne($zero, $t1, there)
```
The parentheses make this a little too dissimilar to the surrounding code for my taste, and I'd like to to this:
```
_bne $zero, $t1, there
```

Would that be possible?

mikeakohn, 2025-01-18:
I totally feel your pain on that one.. got bit by forgetting the delay slot before. I'm not sure about this one.. kind of don't want to make that code more complicated than it is. I'd probably want to do a cleanup in that module (was wanting to make it more C++-like anyway) and I think I'd probably want different syntax than ".macro" to make it different than a normal macro... have to think about it...

russells, 2025-01-20:
If I get time and I can understand the code, I'd be happy to submit a patch for your consideration.  Won't be real soon, though.

mikeakohn, 2025-01-25:
I think if I did this, I'd either want to do a refactor of the macro code or just have a different feature built around it. I'd probably prefer doing that myself...