mikeakohn/naken_asm #142: Macro parsing a character literal containing a comma

russells, 2025-01-30:
Looks like macro parsing of commas happens even inside a character literal.

Best illustrated with an example:

```
$ < t.asm awk '{print NR, "\t", $0}'
1               .pic32
2               li $t0, ','             ; OK
3        
4               .macro m(c)
5               li $t0, c
6               .endm
7        
8               m('A')                  ; OK
9               m(',')                  ; Error
$ naken_asm t.asm

naken_asm

Authors: Michael Kohn
         Joe Davisson
    Web: https://www.mikekohn.net/
  Email: mike@mikekohn.net
Version: October 19, 2024

 Input file: t.asm
Output file: out.hex

Pass 1...
Error: Macro expects 1 params, but got only 2 at t.asm:9.
Pass 2...
Error: Macro expects 1 params, but got only 2 at t.asm:9.

Program Info:
Include Paths: .
               /usr/local/share/naken_asm/include
               include
 Instructions: 2
   Code Bytes: 8
   Data Bytes: 0
  Low Address: 0x0000 (0)
 High Address: 0x0007 (7)

$ 
```

mikeakohn, 2025-01-30:
I'll see if I can look into this tonight.

mikeakohn, 2025-02-01:
Should be fixed with an extra unit test. Let me know how it goes...

russells, 2025-02-02:
Thank you, macro with a single quoted comma works now.

I've discovered a similar behaviour with a single quote.

This works
~~~
li $t0, '\''
~~~
But this doesn't
~~~
m('\'')
~~~

mikeakohn, 2025-02-02:
The macro code has a different file reading code than the token code.. I forgot why I did that but I was hoping to get a chance to clean up that whole thing. I'll look into this and do a quick fix.

mikeakohn, 2025-02-03:
That should take care of '\''. Let me know how that goes.

mikeakohn, 2025-02-09:
I'm going to take that as it's good. Thanks for letting me know and let me know if you find anything else.

closed 2025-02-09 by mikeakohn